rodolfokeesey/Liquid-Handler #11: ways to eliminate unnecessary soldering

jschrier, 2023-02-17:
I've been observing students have trouble with the soldering for making custom wires.  Admittedly this is a pain.  Some ways to eliminate this:

1. Use stepper motors [1](https://www.amazon.com/STEPPERONLINE-Stepper-Bipolar-Connector-compatible/dp/B00PNEQKC0) [2](https://www.amazon.com/dp/B0B8HB5N4S) with [JST or DuPont connectors](https://3dprinting.stackexchange.com/questions/18385/what-is-this-stepper-motor-connector-called) (rather than bare wires) already attached to simplify the process. This will necessitate adding the proper header (if not DuPont) on to the PCB
2. Use [crimpable connectors to JST](https://www.adafruit.com/product/3835) or find [limit switches with JST connector cables attached](https://www.amazon.com/Genmitsu-Micro-Limit-Switches-3020-PRO/dp/B0B3DRXDRL) to eliminate the need to solder these on (or at least make the attachment easier).  Need to add JST headers to PCB
3. Can also use momentary pushbuttons with crimpable contacts[1](https://www.amazon.com/dp/B086KNMJSP) [2](https://www.amazon.com/dp/B0BCGBL7GK) and use the crimpable->JST connectors above , and add PCB boards
4. Take advantage of the [castellated Raspberry Pi Pico pcb](https://www.pcbdirectory.com/community/what-are-castellated-holes-on-a-pcb) to avoid having to solder header pins on to pico and cut pins

(not a concern with the pumps)

jschrier, 2023-02-17:
Partially inspired by these [ClipZin connectors](https://www.newelectronics.co.uk/content/news/farnell-exclusively-stocking-edac-clipzin-connectors-for-raspberry-pi-pico/ ) for RPi Pico. (which are kind of unnecessary for our purposes)

rodolfokeesey, 2023-02-18:
All good ideas! I think with some sourcing we can find pancake steppers with connectors already attached. Alternatively, if sourcing is difficult, we can use 1.8 degree steppers with micro-stepping, instead of the 0.9 degree ones we now use.

Another low effort solution might be to use these kinds of [soldered heat-shrink connectors](https://www.amazon.com/dp/B0B1PSLWWF/ref=redir_mobile_desktop?_encoding=UTF8&aaxitk=7259dd6afe7730ae5eb9310512a86996&content-id=amzn1.sym.7dd77237-72be-4809-b5b5-d553eab7ad9d%3Aamzn1.sym.7dd77237-72be-4809-b5b5-d553eab7ad9d&hsa_cr_id=8620686110901&pd_rd_plhdr=t&pd_rd_r=539d2d50-f235-47fe-ab2d-17bdb9f3e855&pd_rd_w=yXZPf&pd_rd_wg=egbKY&qid=1676687159&ref_=sbx_be_s_sparkle_lsi4d_asin_0_title&sr=1-1-9e67e56a-6f64-441f-a281-df67fc737124). You'd still need to connect the wires, but heat shrinking the wires might be a more accessible approach than soldering.

jschrier, 2023-12-19:
Nice article on solder-less connection alternatives (with pictures) https://hackaday.com/2023/12/18/ask-hackaday-what-do-you-do-when-you-cant-solder/

jschrier, 2024-04-25:
Apropos #4 (take advantage of the castellated PCB on the Pico to avoid having to solder on header pins and then solder those through hole joints).  A helpful [guide from spark fun on soldering castellated PCBs](https://learn.sparkfun.com/tutorials/how-to-solder-castellated-mounting-holes)